UBC-Thunderbots/Software #2236: Improve motion constraints for kickoffs

itsarune, 2021-08-24:
### Description of the task

During Robocup, we quickly created a new Kickoff play that allowed us to keep possession. In the 1-robocup branch, we made  `KickoffChipTactic` and `KickoffMoveTactic` which are both just copies of `KickTactic` and `MoveTactic` without any new functionality. We needed it because we needed to set special motion constraints in `motion_constraint_visitor.cpp` that let the kicker into the enemy half centre circle so that the ball can be kicked. We want a macro (possibly defined in `tactic.h`) that lets us duplicate/extend an existing tactic with a new name without needing to make new header and source files.

For this ticket, just make this macro and use it to make `KickoffChipTactic`, so that we can delete `kickoff_chip_tactic.h` and `kickoff_chip_tactic.cpp`.

Also, in `motion_constraint_set_builder.cpp`, adjust the conditions so that the motion constraint HALF_METER_AROUND_BALL is only active during kickoffs if we are still in the setup state.

### Acceptance criteria

- [ ] Make a macro that lets us duplicate/extend tactics with new names
- [ ] Use this macro to make `KickoffChipTactic` and delete `kickoff_chip_tactic.h` and `kickoff_chip_tactic.cpp`
- [ ] Adjust the HALF_METER_AROUND_BALL motion constraint for kickoffs
- [ ] Make sure that motion constraint behaviour looks correct by looking at some kickoff gameplay on AI vs AI

### Blocked By

BrianS99, 2021-09-25:
Is this macro I am making going to be used for all tactics in the future or is it just to duplicate tactics really quickly during RoboCup?

itsarune, 2021-09-26:
This isn't for all tactics in the future, nor is it for  RoboCup only. The reason we're doing this is because we assign motion constraints to robots based on what tactic they have.
Sometimes we just want a move tactic that can't cross into the enemy half (eg. kickoff), that's why we need to create a copy of move tactic for kickoff (called KickoffMoveTactic) that the system will assign different constraints to

PR #2294 (2022-01-15): Improve Motion Constraints and Setup for Kickoffs

## Please fill out the following before requesting review on this PR

### Description
Deleted `kickoff_chip_tactic.cpp` and `kickoff_chip_tactic.h` and added a macro in `tactic.h` that allows us to duplicate tactic classes. Recreated KickoffChipTactic class using ChipTactic as a copy. Also changed `motion_constraint_set_builder.cpp` so that HALF_METER_AROUND_BALL is only active during kickoffs if we are still in the setup state.

### Testing Done
Reviewed AI vs. AI plays on simulator and it looks normal.

### Resolved Issues
Resolves #2236 

# [... cut]
closed 2022-01-15 by BrianS99