UBC-Thunderbots/Software #2242: Clean up vector of vector of Tactics

jonathanlew, 2021-09-21:
### Description of the task
Last year, we experimented with returning a vector of vector of Tactics from Plays to have more control over priority and reduce oscillations, but it didn't achieve its goals, so it can be removed to simplify things

### Acceptance criteria

- [ ] Replace vector of vector of Tactics with a vector of Tactics

### Blocked By

amrm3lm, 2021-09-22:
wasnt this still helpful for assigning tactcs when our robots had yellow/red cards?

jonathanlew, 2021-09-22:
I don't recall making special behaviours for yellow/red cards. Do you have an example?

amrm3lm, 2021-09-22:
in robocup we changed the order of vectors for `shoot_or_pass` play so that 2 `crease_defenders` were always assigned first. We did this to prioritize `crease_defenders` since not all tactics could be assigned when we were down multiple robots (due to yellow/red cards)

jonathanlew, 2021-09-23:
can't it be implemented without the single layer of vector?

akhilveeraghanta, 2021-09-24:
This tiered assignment thing tried to paper over the underlying assignment issue but it didn't work

If this ticket involves updating STP then its not a 1 story point, i.e `{{tactic1}, {tactic2, tactic3}} -> {tactic1, tactic2, tactic3}`

if its just {{tactic1}, {tactic2, tactic3}} -> {{tactic1, tactic2, tactic3}} then 1 story point makes sense

amrm3lm, 2021-09-25:
I looked at `stp.cpp` again and realized that yes any vector can work fine

jonathanlew, 2021-11-12:
vector of vector is actually useful, so not going to clean this up

closed 2021-11-12 by jonathanlew