UBC-Thunderbots/Software #2566: Field Testing PyTest Framework

jonathanlew, 2022-04-05:
### Description of the task
While we can tell the simulator where to initially position the ball and robots, we cannot do the same with field testing. We'd like to avoid manual robot and ball placement. Automatic robot placement is as easy as giving the robots the right command, and automatic ball placement can be done by dribbling the ball.
The goal with this issue is to add a mode to the simulated pytest, where the initial robot and ball positions are automatically translated into the correct robot and ball placement commands, so tests can run on the simulator or on the field by simply swapping out command line arguments. 
One challenge is dealing with cases where there aren't enough robots on the field to run the test, or cases where the robot ids don't match. In the former case, the test can't run, and in the latter case, the test framework could suggest a different robot id mapping

### Acceptance criteria

- [ ] Automatic robot and ball placement
- [ ] Field Testing PyTest Framework

### Blocked By

nimazareian, 2022-09-29:
Closed #2687 in favor of creating a new PR with the new field testing branch: https://github.com/amrm3lm/Software/tree/amr/robot-diagnostics-2-pytests

nimazareian, 2023-02-16:
Consider adding `@pytest.fixture(scope="session")` to a new Thunderscope test fixture (would likely have to wrap `thunderscope.show()`) to avoid Thunderscope from relaunching after every pytest

itsarune, 2023-12-02:
closed via #2906

closed 2023-12-02 by itsarune