UBC-Thunderbots/Software #2619: Simulated Pytests: Freezes when running in parallel

akhilveeraghanta, 2022-05-19:
### Description of the task

The simulated test fixture has a fixed runtime directory. When running multiple tests in parallel, they conflict and make no progress because they are using the same ports. 

Also, a gamecontroller with a fixed port is spawned for every test, and if tests are running in parallel, one of them will fail to start. 

As a stop gap we can do `tags = ['exclusive']` to run the pytests in series, but we should fix this bug so that when we have more pytests we can run in parallel
 

### Acceptance criteria

- [ ] Update simulated test fixture to chose a runtime directory based on the test case name
- [ ] Update gamecontroller in binary_context_manager.py to find an unused ports and use them (rather than fixed constants)
- [ ] Test that multiple pytests can run in parallel (`bazel test  //software/ai/...` would work, run it once, then change `ball_placement_test.py` and `goalie_tactic_test.py` to make them fire at once) 

### Blocked By

N/A

nimazareian, 2024-05-22:
When we remove the `"exclusive"` tag from tests, qt-material theme throws the following error:
```
self = <qt_material.resources.generate.ResourseGenerator object at 0x7f8bec06fc40>
...
>                   with open(file_to_write, 'w') as file_output:
E                   OSError: [Errno 30] Read-only file system: '/home/nima/.qt_material/theme/disabled/checklist.svg'
```
This is due to Bazel running the tests in a different sandbox environment where the entire filesystem but the files in the sandbox are readonly to allow for tests to be ran in parallel. There's a bit more detail here: https://bazel.build/docs/sandboxing#:~:text=That%20is%2C%20it%20makes%20the%20entire%20filesystem%20read-only%20except%20for%20the%20sandbox%20directory%2C%20so%20the%20action%20cannot%20accidentally%20modify%20anything%20on%20the%20host%20filesystem.%20This%20prevents%20situations%20like%20a%20buggy%20test%20accidentally%20rm%20-rf'ing%20your%20%24HOME%20directory

Furthermore, qt-material attempts to create a couple of icons during runtime and write it to a path outside of the sandbox (in my case `/home/nima/.qt_material`) which is read-only. 

To have Bazel allow qt-material to write to this directory, we can pass the following argument to our bazel command `--sandbox_writable_path /home/nima/.qt_material`. This can likely be added to `.bazelrc` so we don't have to add it manually everytime.

cc: @Mr-Anyone

nimazareian, 2024-05-23:
I looked further into a permanent solution, and it doesn't seem like there is a way of referencing the home directory within the `.bazelrc` file, thus having something like `test --sandbox_writable_path=~/.qt_material` in our `.bazelrc` doesn't work. Another issue is that the `~/.qt_material` folder is created during the first runtime of the package. This means that bazel will throw an error that the `sandbox_writable_path` is invalid.

An alternative solution is to create a folder named `theme` under `/opt/tbotspython` (could be done in `setup_software.sh`), and have qt-material use that folder to store its icons by passing the argument `parent="/opt/tbotspython/theme"` to the `apply_stylesheet` call here: https://github.com/UBC-Thunderbots/Software/blob/1528173facd4b72d7fd4265e99da5ca9e26b1e3d/src/software/thunderscope/thunderscope_config.py#L55

The `.bazelrc` file can then updated to have that folder writable by adding the following line to it:
```
# qt-material theme generates some icons during runtime stored here, thus the path
# will need to be made writable
test --sandbox_writable_path=/opt/tbotspython/theme
```