UBC-Thunderbots/Software #2766: Look into updating the Jetsons to Jetpack 4.6

nimazareian, 2022-10-20:
### Description of the task
Jetpack 4.6 has been released by Nvidia for a while, so we should first check to see what version we're currently running. If it's an older version, we should look into upgrading to 4.6 as it may help with the SPI issues we've been having. https://docs.nvidia.com/jetson/jetpack/install-jetpack/index.html#upgrade-jetpack

Note that Jetpack 5 has been released, but is [not supported](https://forums.developer.nvidia.com/t/jetson-software-roadmap-for-2h-2021-and-2022/177724) on the Jetson nano!

### Acceptance criteria

- [ ] Check current version of Jetpack and upgrade to 4.6 if its older
- [ ] Update to newest Ubuntu 18.04 kernel, if available

### Blocked By

itsarune, 2023-02-25:
4.6 changed how device trees are structured and had some trouble updating it to the correct device_tree

itsarune, 2023-02-25:
4.5 runs on Robot 5, and has some spi performance improvements

nimazareian, 2023-04-27:
Should also update `CONTROL_LOOP_HZ` to match the updated runtime of thunderloop. 

https://github.com/UBC-Thunderbots/Software/blob/dedb5b845b7600a102d7f163dfcd45e4a1f47264/src/shared/constants.h#L162

As part of #2896 I'm reducing the value to 60 so it can run more consistently

itsarune, 2023-05-25:
bruhbot and aimbot have both been upgraded to 4.6.2 (motors spinning)

itsarune, 2023-07-12:
all Jetsons have been upped to 4.6.2

closed 2023-07-12 by itsarune