UBC-Thunderbots/Software #2804: Thunderscope: Verify that Thunderscope launches without error in CI

nimazareian, 2023-01-11:
### Description of the task
Currently, CI does not test that Thunderscope can be launched without an error, thus, we can have a broken Thunderscope be merged with the master branch without the reviewers noticing. 
A simple approach to this would be to test that Thunderscope can be launched in CI and that it can remain running for X seconds. A further improvement would be to check for Python/C++ error messages in that time frame. Prior to trying these approaches, the assignee should research whether other approaches exist for testing PyQt/PyQtGraph applications in Python, as that might simplify the testing and could enrich our ability to test with a greater detail.

### Acceptance criteria

- [ ] If Thunderscope can not be launched or has an error, CI should notice it and let the author and reviewers know.

### Blocked By

itsarune, 2023-07-12:
This environment variable could help in CI: https://stackoverflow.com/a/55446762

By default, Thunderscope will error out on launch in CI

nimazareian, 2024-05-22:
#2788 should now detect most tscope errors

closed 2024-05-22 by nimazareian