UBC-Thunderbots/Software #2855: Thunderloop should call PrimitiveExecutor::stepPrimitive with the actual elapsed time

nimazareian, 2023-03-10:
### Description of the task
Thunderloop currently uses a static time, and if the loop ends up running slower than expected, the primitive executor and HRVO are gonna receive the incorrect delta time

### Acceptance criteria

- [ ] Call stepPrimitive with a dynamic time
- [ ] Remove time_step from PrimitiveExecutor and HRVOSimulator constructor and class

### Blocked By
#2814

itsarune, 2023-12-02:
@nimazareian Is this still necessary for the new motion planner? We can close it if it isn't.

nimazareian, 2023-12-20:
> @nimazareian Is this still necessary for the new motion planner? We can close it if it isn't.

Although the task description is focused around HRVO, it could still be useful to support dynamic delta time for trajectory motion as well