UBC-Thunderbots/Software #2903: HRVO: Dribble backward slower

itsarune, 2023-05-06:
### Description of the task

If we have the ball and we're dribbling backwards, dribble slower

### Acceptance criteria

- [ ] 

### Blocked By

mkhlb, 2024-05-07:
https://www.notion.so/ubcthunderbots/Pivot-Kick-Field-Testing-Notes-3rd-May-2024-b1dca087cef745358100bd737c82cf20

Based on tests outlined here, dribbling sideways was also identified as a weakness of dribbling, along with forwards, but to varying degrees. Perhaps this issue could be instead solved by creating a velocity profile for all directions rather than just changing backwards movement. Alternatively we could solve the sideways issue by just avoiding dribbling sideways in the logic.

itsarune, 2024-06-01:
Possibly covered by #2848

itsarune, 2024-06-01:
We can make __all__ dribbling slower