UBC-Thunderbots/Software #2936: No protos being received after Stop command

williamckha, 2023-06-11:
### Description of the task

Occasionally after a `Stop` command is sent, all robots stop responding. Looks like no protos are received. Likely related to #2932

[image]

### Acceptance criteria

- [ ] 

### Blocked By

williamckha, 2023-06-21:
Closed in favour of #2941

closed 2023-06-21 by williamckha

nimazareian, 2023-06-21:
Next time you see this bug, could you check to see whether yellow fullsystem has also crashed? (Can look at the proto plotter in the Yellow Fullsystem tab)

williamckha, 2023-06-21:
> Next time you see this bug, could you check to see whether yellow fullsystem has also crashed? (Can look at the proto plotter in the Yellow Fullsystem tab)

Yeah yellow fullsystem also crashed