UBC-Thunderbots/Software #2941: SIGABRT (Circle radius cannot be negative) after STOP command issued

williamckha, 2023-06-21:
### Description of the task

I got a SIGABRT immediately after issuing the STOP command, which caused all robots to hang. I've been unable to reproduce this SIGABRT a second time

```
terminate called after throwing an instance of 'std::invalid_argument'
  what():  Circle radius cannot be negative, given: -0.662326
```

Full logs:

```
2023/06/20 17:24:43 Engine: Process change '{"origin":"UI","revertible":true,"newCommand":{"command":{"type":"STOP","forTeam":"UNKNOWN"}}}'
2023/06/20 17:24:43 Processing change '{"origin":"UI","revertible":true,"newCommand":{"command":{"type":"STOP","forTeam":"UNKNOWN"}}}'
2023/06/20 17:24:43 Change '{"origin":"UI","revertible":true,"newCommand":{"command":{"type":"STOP","forTeam":"UNKNOWN"}}}' processed:
from: {"stage":"NORMAL_FIRST_HALF","command":{"type":"DIRECT","forTeam":"BLUE"},"gameState":{"type":"RUNNING"},"stageTimeElapsed":"305.742341495s","stageTimeLeft":"-5.742341495s","matchTimeStart":"2023-06-21T00:07:57.158117571Z","teamState":{"BLUE":{"name":"Unknown","goals":0,"goalkeeper":0,"timeoutsLeft":4,"timeoutTimeLeft":"300s","onPositiveHalf":false,"ballPlacementFailures":0,"ballPlacementFailuresReached":false,"canPlaceBall":true,"maxAllowedBots":11,"challengeFlags":3},"YELLOW":{"name":"Unknown","goals":0,"goalkeeper":0,"timeoutsLeft":4,"timeoutTimeLeft":"300s","onPositiveHalf":true,"ballPlacementFailures":0,"ballPlacementFailuresReached":false,"canPlaceBall":true,"maxAllowedBots":11,"challengeFlags":3}},"currentActionTimeRemaining":"-9.223918660s","division":"DIV_A","autoContinue":true,"firstKickoffTeam":"YELLOW","matchType":"UNKNOWN_MATCH"}
  to: {"stage":"NORMAL_FIRST_HALF","command":{"type":"STOP","forTeam":"UNKNOWN"},"gameState":{"type":"STOP"},"stageTimeElapsed":"305.742341495s","stageTimeLeft":"-5.742341495s","matchTimeStart":"2023-06-21T00:07:57.158117571Z","teamState":{"BLUE":{"name":"Unknown","goals":0,"goalkeeper":0,"timeoutsLeft":4,"timeoutTimeLeft":"300s","onPositiveHalf":false,"ballPlacementFailures":0,"ballPlacementFailuresReached":false,"canPlaceBall":true,"maxAllowedBots":11,"challengeFlags":3},"YELLOW":{"name":"Unknown","goals":0,"goalkeeper":0,"timeoutsLeft":4,"timeoutTimeLeft":"300s","onPositiveHalf":true,"ballPlacementFailures":0,"ballPlacementFailuresReached":false,"canPlaceBall":true,"maxAllowedBots":11,"challengeFlags":3}},"currentActionTimeRemaining":"-9.223918660s","division":"DIV_A","autoContinue":true,"firstKickoffTeam":"YELLOW","matchType":"UNKNOWN_MATCH"}
2023/06/20 17:24:43 Add entry to state store
2023/06/20 17:24:43 Notify hook  messageGen
2023/06/20 17:24:43 Notify hook  apiServer-fd3b8bfd-02af-4237-b544-6b433983d66d
2023/06/20 17:24:43 Change produced 0 new changes
2023/06/20 17:24:43 Change '{"origin":"UI","revertible":true,"newCommand":{"command":{"type":"STOP","forTeam":"UNKNOWN"}}}' processed
terminate called after throwing an instance of 'std::invalid_argument'
  what():  Circle radius cannot be negative, given: -0.662326
2023/06/20 17:24:43 953767

***** FATAL SIGNAL RECEIVED ******* 
Received fatal signal: SIGABRT(6)       PID: 18460

***** SIGNAL SIGABRT(6)

******* STACKDUMP *******
        stack dump [1]  software/er_force_simulator_main+0x33f2c0 [0x5601b773c2c0]
        stack dump [2]  /lib/x86_64-linux-gnu/libpthread.so.0+0x14420 [0x7f06eb509420]
        stack dump [3]  /lib/x86_64-linux-gnu/libc.so.6gsignal+0xcb [0x7f06eb1e500b]
        stack dump [4]  /lib/x86_64-linux-gnu/libc.so.6abort+0x12b [0x7f06eb1c4859]
        stack dump [5]  software/er_force_simulator_main+0x68f3a [0x5601b7465f3a]
        stack dump [6]  software/er_force_simulator_main+0x5e455c [0x5601b79e155c]
        stack dump [7]  software/er_force_simulator_main+0x5e45c7 [0x5601b79e15c7]
        stack dump [8]  software/er_force_simulator_main+0x5e4729 [0x5601b79e1729]
        stack dump [9]  software/er_force_simulator_main+0x2c88c4 [0x5601b76c58c4]
        stack dump [10]  software/er_force_simulator_main+0x2641d0 [0x5601b76611d0]
        stack dump [11]  software/er_force_simulator_main+0x26ba54 [0x5601b7668a54]
        stack dump [12]  software/er_force_simulator_main+0x26c7e7 [0x5601b76697e7]
        stack dump [13]  software/er_force_simulator_main+0x253f4f [0x5601b7650f4f]
        stack dump [14]  software/er_force_simulator_main+0x24f59c [0x5601b764c59c]
        stack dump [15]  software/er_force_simulator_main+0xb0730 [0x5601b74ad730]
        stack dump [16]  software/er_force_simulator_main+0xb0e68 [0x5601b74ade68]
        stack dump [17]  software/er_force_simulator_main+0x6c75d [0x5601b746975d]
        stack dump [18]  software/er_force_simulator_main+0x6ea5e [0x5601b746ba5e]
        stack dump [19]  software/er_force_simulator_main+0x94e9f [0x5601b7491e9f]
        stack dump [20]  software/er_force_simulator_main+0x91430 [0x5601b748e430]
        stack dump [21]  software/er_force_simulator_main+0xa4466 [0x5601b74a1466]
        stack dump [22]  software/er_force_simulator_main+0xa2f55 [0x5601b749ff55]
        stack dump [23]  software/er_force_simulator_main+0xa1a49 [0x5601b749ea49]
        stack dump [24]  software/er_force_simulator_main+0xa0861 [0x5601b749d861]
        stack dump [25]  software/er_force_simulator_main+0x9fa8b [0x5601b749ca8b]
        stack dump [26]  software/er_force_simulator_main+0x9eb45 [0x5601b749bb45]
        stack dump [27]  software/er_force_simulator_main+0x9d7b9 [0x5601b749a7b9]
        stack dump [28]  software/er_force_simulator_main+0x9bbc3 [0x5601b7498bc3]
        stack dump [29]  software/er_force_simulator_main+0x745ca [0x5601b74715ca]
        stack dump [30]  software/er_force_simulator_main+0x781af [0x5601b74751af]
        stack dump [31]  software/er_force_simulator_main+0x745ca [0x5601b74715ca]
        stack dump [32]  software/er_force_simulator_main+0x78e5e [0x5601b7475e5e]
        stack dump [33]  software/er_force_simulator_main+0x78873 [0x5601b7475873]
        stack dump [34]  software/er_force_simulator_main+0x7934c [0x5601b747634c]
        stack dump [35]  software/er_force_simulator_main+0x81781 [0x5601b747e781]
        stack dump [36]  software/er_force_simulator_main+0xac9eb [0x5601b74a99eb]
        stack dump [37]  software/er_force_simulator_main+0xab59b [0x5601b74a859b]
        stack dump [38]  software/er_force_simulator_main+0xaa656 [0x5601b74a7656]
[... 54 lines cut ...]
        stack dump [38]  software/er_force_simulator_main+0xaa656 [0x5601b74a7656]
        stack dump [39]  software/er_force_simulator_main+0xa9622 [0x5601b74a6622]
        stack dump [40]  software/er_force_simulator_main+0xa75fc [0x5601b74a45fc]
        stack dump [41]  software/er_force_simulator_main+0x668724 [0x5601b7a65724]
        stack dump [42]  /lib/x86_64-linux-gnu/libpthread.so.0+0x8609 [0x7f06eb4fd609]
        stack dump [43]  /lib/x86_64-linux-gnu/libc.so.6clone+0x43 [0x7f06eb2c1133]

Exiting after fatal event  (FATAL_SIGNAL). Fatal type:  SIGABRT
Log content flushed sucessfully to sink

 
Exiting, log location: /tmp/tbots/sim/thunderbots.log

Exiting, log location: /tmp/tbots/sim/thunderbots_filtered.log

Exiting, log location: /tmp/tbots/sim/thunderbots_text.log

exitWithDefaultSignalHandler:229. Exiting due to FATAL_SIGNAL, 6  
```

### Acceptance criteria

- [ ] 

### Blocked By

nimazareian, 2023-06-21:
I've also noticed this multiple times, but haven't been able to reproduce it under a debugger to know where its from. I suspect that it's related to HRVO, but looking at the code, I haven't found anything that could give a negative radius. From my observations, the negative radius can be pretty large. It's likely that it is an uninitialized variable with a random value. 
I've noticed it happen when robots are moving fast and they collide.

itsarune, 2023-06-21:
really? The stack dump looks er_force_sim related no?

williamckha, 2023-06-21:
This might be the culprit? If the velocity gets obscenely large 

https://github.com/UBC-Thunderbots/Software/blob/49474f344071bb0587847739704554694e92905c/src/software/ai/navigator/path_planner/hrvo/agent.cpp#L73-L83

nimazareian, 2023-06-22:
> This might be the culprit? If the velocity gets obscenely large

Yeah I guess it could be, it's worth clamping it. I've opened a PR for it #2943

> really? The stack dump looks er_force_sim related no?

Well the stack dump doesn't really give us much info (atleast that I could understand). And it's worth noting that in simulation, HRVO runs in the Er Force Sim process...

nimazareian, 2023-08-28:
@williamckha @itsarune Have you noticed this segfault anymore? I haven't noticed it since William's suggestion in #2943. Just want to see if we should close it.

itsarune, 2023-08-28:
We should be good to close, I haven't seen it since