UBC-Thunderbots/Software #3004: Thunderscope: Add FPS counter for field widget

nimazareian, 2023-10-02:
### Description of the task
We should add a FPS counter for the field widget to show how fast Thunderscope is rendering. This way we can notice how efficient Thunderscope is running.

### Acceptance criteria

- [ ] 

### Blocked By

nimazareian, 2023-12-12:
It could be interesting to also plot the rate at which we send out primitiveSets to the robots, so we can tell if our Python code can not keep up with our desired frequency 

https://github.com/UBC-Thunderbots/Software/blob/7a9cba4948887c91293318fbc49a5422521ae940/src/software/thunderscope/robot_communication.py#L223-L241

https://github.com/UBC-Thunderbots/Software/blob/7a9cba4948887c91293318fbc49a5422521ae940/src/software/thunderscope/robot_communication.py#L293-L294

nimazareian, 2023-12-12:
Likewise, I think we should measure the rate at which primitivesets are received at on the robot as well and put it in the `RobotStatus` protobuf message. Very often do we have problems where it seems like the robot isn't being responsive, and it could be due to a lack of primitivesets being received (e.g. due to packet loss or Python bottleneck not sending packets ...)

Mr-Anyone, 2024-02-10:
I am unsure whether to change field widget to QOpenGLWidget to use framebuffer swap.

nimazareian, 2024-02-10:
> I am unsure whether to change field widget to QOpenGLWidget to use framebuffer swap to measure FPS.

You should be able to connect to the framebuffer swap signal in `ExtendedGLViewWidget` as that extends `GLViewWidget` which extends `QtWidgets.QOpenGLWidget`

https://doc.qt.io/qt-6/qopenglwidget.html#frameSwapped

https://github.com/UBC-Thunderbots/Software/blob/dfb13b53b82c0628b03132c7042e7bee7e1397be/src/software/thunderscope/gl/helpers/extended_gl_view_widget.py#L36-L38

Mr-Anyone, 2024-02-10:
> You should be able to connect to the framebuffer swap signal in `ExtendedGLViewWidget` as that extends `GLViewWidget` which extends `QtWidgets.QOpenGLWidget` [...]

I see. Will check this out.

nimazareian, 2024-03-30:
Closed in #3125

closed 2024-03-30 by nimazareian