UBC-Thunderbots/Software #3012: Support variable chip power

nimazareian, 2023-10-06:
### Description of the task
Similar to the variable kick power, we should support variable chip power. 

### Acceptance criteria

- [ ] Support variable chip power
- [ ] Collaborate with mech to tune the pulse_width to chip distance relationship
- [ ] Test this using Robot Diagnostics
- [ ] Update comment in Robot Diagnostics that power is constant for chipping

### Blocked By

nimazareian, 2023-10-06:
#2945 includes detail on the updated relationship we use for variable kick power.

nimazareian, 2023-11-03:
Update this line so its not constant:

https://github.com/UBC-Thunderbots/Software/blob/88bb2f0d3c63549a8991b1aac4ed939fd0e7cf00/src/proto/message_translation/power_frame_msg.hpp#L143

Update redis so it reads the robot specific constant:

https://github.com/UBC-Thunderbots/Software/blob/88bb2f0d3c63549a8991b1aac4ed939fd0e7cf00/src/software/jetson_nano/thunderloop.cpp#L79-L80

Shouldn't need to update the `Primitive` protobuf that we send from AI/diagnostics to the robot as it already includes a chip distance field

https://github.com/UBC-Thunderbots/Software/blob/88bb2f0d3c63549a8991b1aac4ed939fd0e7cf00/src/proto/primitive.proto#L78-L85

Right now robot diagnostics UI uses the same `power` value for manual + auto chip and kick. We should probably update the UI to have another slider for the chip distance:

https://github.com/UBC-Thunderbots/Software/blob/88bb2f0d3c63549a8991b1aac4ed939fd0e7cf00/src/software/thunderscope/robot_diagnostics/chicker_widget.py#L177-L187

Here's how the UI element for the kick speed slider is created. Could update the title of this, and create a similar one for chip distance:

https://github.com/UBC-Thunderbots/Software/blob/88bb2f0d3c63549a8991b1aac4ed939fd0e7cf00/src/software/thunderscope/robot_diagnostics/chicker_widget.py#L96-L103

ds570001, 2024-01-29:
Will likely need to change the poll statement to include additional chip constants (similar to kick)

https://github.com/UBC-Thunderbots/Software/blob/88bb2f0d3c63549a8991b1aac4ed939fd0e7cf00/src/software/jetson_nano/thunderloop.cpp#L282-L283

ds570001, 2024-02-10:
Thanks to Jeanette / James / Omri, the relation between chip distance (cm) and pulsewidth (microseconds) is as follows:
chip distance = chip_slope * (pulsewidth) + chip_constant

Data: https://docs.google.com/spreadsheets/d/1kKtib3wTgQtTu_EwKyxhcdTrjjLTpx03YDrG7tcgK-w/edit#gid=0
*NOTE* This data is subject to change as the chipper design is currently being redesigned

ds570001, 2024-02-10:
Also realized that the kick - pulsewidth calculation is incorrect. Currently the calculation is pulse width = a*exp(b*kick_speed)

https://github.com/UBC-Thunderbots/Software/blob/dfb13b53b82c0628b03132c7042e7bee7e1397be/src/proto/message_translation/power_frame_msg.hpp#L141-L144

According to the data found in https://github.com/UBC-Thunderbots/Software/pull/2945, the exponential function takes *pulse width* as input and *kick speed* as output
[image]

So the kick code should be updated so it takes the inverse of the exponential relation

ds570001, 2024-02-10:
- [ ] fix the kick code so that it uses the inverse relationship
- [ ] double check that the constants are correct on the robots

The inverse relationship should be ln(kick_speed / kick_constant) / kick_coeff

nimazareian, 2024-02-10:
We might be able to fit a quadratic function now as well.

ds570001, 2024-02-10:
@nimazareian I think we should change the safety mechanism so it selects the minimum of the computed pulse width and a max width constant, instead of selecting the minimum between equation coefficients. We calculated the max safety coefficients by looking at the max values of each coefficient from the data, as shown below, but this ends up producing a super large pulse width relative to the max pulse width we see from the data:
[image]

if we instead focused on the max pulse width from the data, we could set it to 6000 for example (much lower, and less of a chance that the power board will break). The reason i say this is because we saw recently that when mech was changing the solenoids they started breaking power boards because of the different solenoid construction and we were pulsing them too long when we send a 10000 pulse width.

Here's the code where the safety constraints is implemented:

https://github.com/UBC-Thunderbots/Software/blob/dfb13b53b82c0628b03132c7042e7bee7e1397be/src/shared/constants.h#L138-L139

https://github.com/UBC-Thunderbots/Software/blob/dfb13b53b82c0628b03132c7042e7bee7e1397be/src/proto/message_translation/power_frame_msg.hpp#L132-L134