UBC-Thunderbots/Software #3031: Robot Automatic Subtitute FSM

jameschuang2002, 2023-10-14:
### Description of the task

Note: We might want to separate this ticket into multiple tickets. One for each faulty behaviour and another one for the FSM design.

There should also be a function that reads from all the fault detection functions and return whether a substitute is needed. The FSM will only be implemented when this function returns true. The designer of the first fault detection function should implement this function. 

Inspired by Tiger, we would like to implement a auto-substituting mechanism when detecting scenarios that the robots were displaying faulting behaviours. The FSM, when detect an error, will move the robot to the sidelines during the game so the team could place another robot into the game. 

Faulty Behaviours Includes: (Please feel free to add to this list of faulty behaviours)
1. The robot believes it is dribbling the ball when the ball is actually somewhere else

### Acceptance criteria

Please feel free to change it if relevant. 

FSM Requirements 
- [ ] State transition as expected
- [ ] Robot that should be substituted moved to the sideline automatically 
- [ ] Tests that verify should/shouldn't substitute situations

Faulty Detection Function Requirements
- [ ] Ability to detect the fault
- [ ] Comprehensive test suite that tested regular and edge cases for the fault. 

### Blocked By

itsarune, 2023-10-17:
2. If the motor status messages have faults in them

Also we should be able to enable/disable this feature through dynamic parameters in case we don't have spare robots.

itsarune, 2023-10-17:
Right now, as I think about this, one challenging part is that we don't have a way to fix certain robots to tactic ids