UBC-Thunderbots/Software #3047: [Robot Software] Add CPU and Memory usage of the Jetsons to the robot status msg

nimazareian, 2023-10-30:
### Description of the task
We should consider logging the CPU and Memory usage of the Jetsons through the robot status msg. This could possibly help us figure out if we've added some faulty code that is throttling the Jetson. We can then add alerts for if these spike up above a scenario. Similarly, we could add Thunderscope alerts if the thunderloop frequency goes below some threshold.

Should investigate that whatever method is used to measure these statistics does not actually hurt our performance. I've read that at some point in time `top` produced misleading stats, and using something like `trace-cmd` mightve been more accurate (though I havent actually researched the difference myself).

### Acceptance criteria

- [ ] 

### Blocked By

itsarune, 2023-10-31:
I feel like CPU and memory spikes are a bit random and its hard to correlate it to the code we're writing

Maybe a better way is to see if thunderloop is failing to keep up with the target frequency that we're setting and seeing if it is consistently happenning

nimazareian, 2023-10-31:
Ya I agree that the best alert we can probably add is related to thunderloops frequency. We should see if there are other useful alerts/indicators we want to add (maybe packet loss %?). With @sauravbanna  's new widget it should be pretty straightforward.