UBC-Thunderbots/Software #3116: Extra robots should be removed from AI vs AI when a team receives a foul

nimazareian, 2024-01-20:
### Description of the task
When a team incurres a foul, the SSL rules require the team to remove 1 robot for 30 seconds. When using AutoRef in simulation, we should detect such events from the game controller and automatically remove a robot from the field so the game controller can continue. To simplify the logic, this can be any robot (but the goalie, unless it's the last robot remaining lol). Ideally this logic should not be tied to the autoref, and should only depend on the msgs from the game controller.

Here's an example msg that the game controller shows when the yellow team has too many robots
```
{
  "gameEvent": {
    "type": "TOO_MANY_ROBOTS",
    "origin": [
      "Engine"
    ],
    "tooManyRobots": {
      "byTeam": "YELLOW",
      "numRobotsAllowed": 5,
      "numRobotsOnField": 6,
      "ballLocation": {
        "x": -0.23220448,
        "y": 0.59790707
      }
    }
  }
}
```
An example of how this could be done can be found in #3033 in `GLSandboxWorldLayer.__remove_robot_from_state` and its surrounding implementation.

### Acceptance criteria

- [ ] Remove extra robots from field when commanded by the gamecontroller/autoref
- [ ] continue the game afterwards

### Blocked By

itsarune, 2024-05-18:
`./tbots.py run thunderscope_main -o --enable_autoref`

nimazareian, 2024-06-22:
After further research, the challenge with using `GLSandboxWorldLayer.__remove_robot_from_state` is that `GLSandboxWorldLayer.refresh_graphics` is only called when the corresponding fullsystem is visible, so we can't rely on it.

I've written an alternative implementation as part of `GameController` class here. But this implementation is unstable, since the behaviour of the GC and the autoref seem somewhat unpredictable as well. More research will need to be put into this to iron out the kinks. I will probably stop working on this for now. Here's my WIP branch:  https://github.com/nimazareian/Software/tree/nima/autoref_remove_extra_robots

closed 2025-06-06 by Andrewyx