UBC-Thunderbots/Software #3244: [Thunderscope] Improve ball_is_off_ground.py Validation Test Visual

Andrewyx, 2024-07-06:
### Description of the task

The current `ball_is_off_ground.py` Validation PyTest uses a flat 2D circle around the ball as its only visualization indicator. The test checks for whenever the ball is off the ground (eg. used when we test for chipping), and should have a better visual for this. 

Most of our current tests use 2D shapes to indicate when conditions such as when the ball enters a region occur or if a robot moves the ball. Since chipping occurs in the z-axis, we need a more creative implementation of current tools to achieve an effective visualization. Below is the folder containing commonly used geometry.
https://github.com/UBC-Thunderbots/Software/tree/master/src/software/geom

### Acceptance criteria

- [ ] An intuitive visual is made
- [ ] Thunderscope runs
- [ ] All simulated tests pass

### Blocked By

nycrat, 2025-10-04:
@Andrewyx I was talking with william about this issue and he suggested to make a 3D visualization like an arrow pointing up at the ball (my original idea was just a 2D arrow in the direction the ball is moving/getting chipped). I was  wondering if we want to change some logic in the validation visualization layer for thunderscope to accommodate for 3D shapes or just stick with a 2D visualization?

Andrewyx, 2025-10-05:
> [@Andrewyx](https://github.com/Andrewyx) I was talking with william about this issue and he suggested to make a 3D visualization like an arrow pointing up at the ball (my original idea was just a 2D [...]

Interesting idea! Do you have a sketch or something to draft out a design? I'm not quite sure I understand exactly how the arrow will look.

nycrat, 2025-10-06:
What I currently have is a 2D arrow pointing in the direction of ball's velocity, with the EventuallyOffGround test, it indicates where and which direction the ball is chipped:

[image]

Some other ideas here for 3D (the top two aren't really possible right now since the validation geometry stops updating once the validation is true once for EventuallyTrue):

[image]