UBC-Thunderbots/Software #3248: Update Thunderloop's `isPowerStable` to work on Pis

nimazareian, 2024-07-09:
### Description of the task
read from `/usr/bin/dmesg` instead. 

### Acceptance criteria

- [ ] 

### Blocked By

nimazareian, 2024-07-09:
@mkhlb Maybe you can include this change in your Pi/Nano Thunderloop changes.

Mr-Anyone, 2024-07-10:
I haven't looked into this, but I think you might also need to look into the following after I grep the project: 

https://github.com/UBC-Thunderbots/Software/blob/8e45ea06e21b98fd86f7355fedfcb38533a76589/src/software/jetson_nano/thunderloop.h#L140

This is used here: 

I am not sure if the rapsberry pi has this file? 

https://github.com/UBC-Thunderbots/Software/blob/8e45ea06e21b98fd86f7355fedfcb38533a76589/src/software/jetson_nano/thunderloop.cpp#L398-L418

Mr-Anyone, 2024-07-10:
Also, do we need to update the `GPIO` class?

nimazareian, 2024-07-10:
Yes you're right about the CPU temperature. I believe when I looked into this a few weeks ago, the Pi does have that file, but it returns a constant and inaccurate value. The correct way to check the value is using `/opt/vc/bin/vcgencmd measure_temp`

Regarding the GPIO class, @mkhlb is looking into refactoring that and Motor service so we can use compilation flags to compile for either of the setups.

closed 2026-05-10 by GrayHoang