UBC-Thunderbots/Software #3294: Add `NotImplementedError` for functions that'll be overridden in Python

itsarune, 2024-07-29:
### Description of the task

To prevent functions that we do not expect to be overridden to be executed in Python, we should raise a `NotImplementedError`

Here are the places (that I know about) where we should add this:
- https://github.com/UBC-Thunderbots/Software/blob/3cc5a2d4c45205f3b0dc7eae6e85f328e2973160/src/software/thunderscope/gl/widgets/gl_toolbar.py#L31

### Acceptance criteria

- [ ] Replace known instances of these abstract function with an implementation that raises a `NotImplementedError`

### Blocked By

Mr-Anyone, 2024-07-29:
I am pretty sure that is something called an abstract base class that  enforces this in python.

williamckha, 2024-07-30:
> I am pretty sure there is something called an abstract base class that enforces this in python.

Yep https://docs.python.org/3/library/abc.html

PR #3378 (2024-11-13): Raised NotImplementedError for base classes without their own impleme…

## Please fill out the following before requesting review on this PR

### Description

Tagged necessary super-classes with abstract method so not implemented error is raised when base class did not implement the method. I went ahead and added this to all the super classes that had methods that needed to be implemented by child classes. (gl_toolbar, tbots_test_runner, gl_shape, gl_layer, validation, time_provider. )

 
### Testing Done

Commented out implementation of child class and ran the build to see if the not implemen [... cut]
closed 2024-11-13 by annieisawesome2