UBC-Thunderbots/Software #3352: Update Hedronvision Compile Commands

Mr-Anyone, 2024-10-12:
### Description of the task

Update a module that generate `compile_commands.json` to make neovim/vim user's life easier. 

### Acceptance criteria

updated and tested the module. 

- [ ] 

### Blocked By

Mr-Anyone, 2025-11-14:
I think there is a bit of weirdness going on: 

```bash 
vhe@vhe:~/vscode/Software/src$ bazel run :refresh_compile_commands
INFO: Invocation ID: b6ba3594-8761-40ab-954e-a5d5fb60b237
WARNING: Download from https://github.com/hedronvision/bazel-compile-commands-extractor/archive/3f28899228fb3ad0126897876f147ca15026151e.tar.gz failed: class java.io.FileNotFoundException GET returned 404 Not Found
INFO: Repository hedron_compile_commands+ instantiated at:
  <builtin>: in <toplevel>
Repository rule http_archive defined at:
  /home/vhe/.cache/bazel/_bazel_vhe/917dd48f905a75cc113dac63db2f3aa5/external/bazel_tools/tools/build_defs/repo/http.bzl:392:31: in <toplevel>
ERROR: /home/vhe/.cache/bazel/_bazel_vhe/917dd48f905a75cc113dac63db2f3aa5/external/bazel_tools/tools/build_defs/repo/http.bzl:137:45: An error occurred during the fetch of repository 'hedron_compile_commands+':
   Traceback (most recent call last):
	File "/home/vhe/.cache/bazel/_bazel_vhe/917dd48f905a75cc113dac63db2f3aa5/external/bazel_tools/tools/build_defs/repo/http.bzl", line 137, column 45, in _http_archive_impl
		download_info = ctx.download_and_extract(
Error in download_and_extract: java.io.IOException: Error downloading [https://github.com/hedronvision/bazel-compile-commands-extractor/archive/3f28899228fb3ad0126897876f147ca15026151e.tar.gz] to /home/vhe/.cache/bazel/_bazel_vhe/917dd48f905a75cc113dac63db2f3aa5/external/hedron_compile_commands+/temp16334146504573877463/3f28899228fb3ad0126897876f147ca15026151e.tar.gz: GET returned 404 Not Found
ERROR: Error computing the main repository mapping: error during computation of main repo mapping: java.io.IOException: Error downloading [https://github.com/hedronvision/bazel-compile-commands-extractor/archive/3f28899228fb3ad0126897876f147ca15026151e.tar.gz] to /home/vhe/.cache/bazel/_bazel_vhe/917dd48f905a75cc113dac63db2f3aa5/external/hedron_compile_commands+/temp16334146504573877463/3f28899228fb3ad0126897876f147ca15026151e.tar.gz: GET returned 404 Not Found
Computing main repo mapping:
```

Can you try on your end @williamckha ?

williamckha, 2025-11-15:
Worked for me... (are you on latest master with Bazel 8 changes?)

```
william@nautilus ~/Software/src (master)> bazel run :refresh_compile_commands
Starting local Bazel server (8.2.1) and connecting to it...
INFO: Invocation ID: 7c31af6c-e492-4a01-b1c7-9bc85115a80b
INFO: Analyzed target //:refresh_compile_commands (96 packages loaded, 1435 targets configured).
INFO: Found 1 target...
Target //:refresh_compile_commands up-to-date:
  bazel-bin/refresh_compile_commands
  bazel-bin/refresh_compile_commands.check_python_version.py
  bazel-bin/refresh_compile_commands.py
INFO: Elapsed time: 3.671s, Critical Path: 0.48s
INFO: 9 processes: 7 internal, 2 processwrapper-sandbox.
INFO: Build completed successfully, 9 total actions
INFO: Running command line: bazel-bin/refresh_compile_commands
>>> Automatically added entries to .git/info/exclude to gitignore generated output.
>>> Automatically added //external workspace link:
    This link makes it easy for you--and for build tooling--to see the external dependencies you bring in. It also makes your source tree have the same directory structure as the build sandbox.
    It's a win/win: It's easier for you to browse the code you use, and it eliminates whole categories of edge cases for build tooling.
>>> Analyzing commands used in //software/...
WARNING: Build options --features and --host_features have changed, discarding analysis cache (this can be expensive, see https://bazel.build/advanced/performance/iteration-speed).
>>> A source file you compile doesn't (yet) exist: bazel-out/k8-fastbuild/bin/proto/tbots_nanopb_proto_nanopb_gen/proto/ball.nanopb.c
    It's probably a generated file, and you haven't yet run a build to generate it.
    That's OK; your code doesn't even have to compile for this tool to work.
    If you can, though, you might want to run a build of your code with --keep_going.
        That way everything possible is generated, browsable and indexed for autocomplete.
    However, if you have *already* built your code, and generated the missing file...
        Please make sure you're supplying this tool with the same flags you use to build.
        You can either use a refresh_compile_commands rule or the special -- syntax. Please see the README.
        [Supplying flags normally won't work. That just causes this tool to be built with those flags.]
    Continuing gracefully...
>>> Finished extracting commands for //software/...
>>> Analyzing commands used in //proto/...
>>> Finished extracting commands for //proto/...
>>> Analyzing commands used in //extlibs/...
>>> Finished extracting commands for //extlibs/...
>>> Analyzing commands used in //shared/...
>>> Finished extracting commands for //shared/...
```