UBC-Thunderbots/Software #3377: Investigate whether using custom acceleration and max velocity parameters for the goalie improves its performance

itsarune, 2024-10-19:
### Description of the task

Currently, all of our robots share acceleration and max velocity constraints (found [here](https://github.com/UBC-Thunderbots/Software/blob/master/src/shared/2021_robot_constants.cpp)).

It may make sense to increase these values for the goalie robot to improve its goalkeeping ability.

Approach:
1. Create [some field tests](https://github.com/UBC-Thunderbots/Software/tree/master/src/software/field_tests) for the goalie robot
2. Evaluate whether higher acceleration and/or higher max velocities improve goalkeeping performance
3. If it does, create new [`RobotConstants`](https://github.com/UBC-Thunderbots/Software/blob/327c96966f667f730ef277fc487da3e074320aae/src/shared/robot_constants.h#L6) with the updated max velocity and acceleration constraints for the goalie.
4. Update [`RobotFilter`](https://github.com/UBC-Thunderbots/Software/blob/master/src/software/sensor_fusion/filter/robot_filter.h) to keep track of the `RobotConstants` to use for each robot. The goalie robot should have the goalie `RobotConstants`.
5. We should associate the `RobotConstants` to each robot through [`SensorFusion`](https://github.com/UBC-Thunderbots/Software/blob/327c96966f667f730ef277fc487da3e074320aae/src/software/sensor_fusion/sensor_fusion.cpp#L63) by updating `friendly_team_filter` and the `robot_filter`.

### Acceptance criteria

- [ ] Add [field tests](https://github.com/UBC-Thunderbots/Software/tree/master/src/software/field_tests) for the goalie robot
- [ ] Test if goalie performs better with higher acceleration and max velocity
- [ ] If per-tactic acceleration and velocity constraints are beneficial, implement a way of setting `RobotConstants` for robots when they are initialized in `SensorFusion`
- [ ] If per-tactic acceleration and velocity constraints are beneficial, add unit tests to [`sensor_fusion_test.cpp`](https://github.com/UBC-Thunderbots/Software/blob/master/src/software/sensor_fusion/sensor_fusion_test.cpp) for the new behaviour introduced for installing new `RobotConstants` for goalie robots

### Blocked By

mkhlb, 2024-10-23:
potentially have a movement mode that cares more about speed than accuracy for things like making blocks or taking close pivot shots

zuperzane, 2025-06-07:
@mkhlb Maybe it could be like the function of BALLSTEALMODE cause thats just kind of sitting there we could use it