UBC-Thunderbots/Software #3487: Floating Point Comparison Issues

Mr-Anyone, 2025-08-25:
https://github.com/UBC-Thunderbots/Software/blob/bc5e7209c86d80ad6f4ac09fd756c3139b440be9/src/software/ai/navigator/obstacle/trajectory_obstacle.hpp#L43

https://github.com/UBC-Thunderbots/Software/blob/bc5e7209c86d80ad6f4ac09fd756c3139b440be9/src/software/ai/navigator/obstacle/trajectory_obstacle.hpp#L59

https://github.com/UBC-Thunderbots/Software/blob/bc5e7209c86d80ad6f4ac09fd756c3139b440be9/src/software/ai/navigator/obstacle/trajectory_obstacle.hpp#L76

https://github.com/UBC-Thunderbots/Software/blob/bc5e7209c86d80ad6f4ac09fd756c3139b440be9/src/software/ai/navigator/obstacle/trajectory_obstacle.hpp#L93

This branch is always false because of floating point comparison? It should probably be changed into following or something along the line: 

```
t_sec < 1e-6
```

Mr-Anyone, 2025-08-25:
Also, this class is really similar to the following class: 

https://mr-anyone.github.io/Thunderbot_Software/classConstVelocityObstacle.html

itsarune, 2025-08-25:
Why isn't there a compiler flag to check this?

itsarune, 2025-08-25:
There is: `-Wfloat-equal`, we should also enable this on our code